ibanezchen/hyperCOS-evaluate #37: For another hardware plateform?

starnight, 2016-06-21:
I want to try the hyperC OS on STM32F4 Discovery.

What header files should I include?  What source files shuold I compile and link?
The most basic files that a developer needs is important.
And, if those information could be found in Wiki, that will be useful.

Thank you ~

ibanezchen, 2016-06-22:
Would you like to post this on the mailing list?
Maybe we can get more people involved.

starnight, 2016-06-22:
That's okay!

ibanezchen, 2016-08-19:
Just push a template and linking script to bootC, branch stm32f429

ibanezchen, 2016-09-07:
Add libstm32f4xx 
todo: 
- bootC 
- tmr_delay
- tmr_interrupt 
- wiki

ibanezchen, 2016-09-12:
libsoc
commit 7b68f608b883a5c073870697babdd55690d62ced
Author: Howard Chen <ibanezchen@gmail.com>
Date:   Mon Sep 12 12:36:39 2016 +0800

    add target stm32f4

closed 2016-09-12 by ibanezchen